beagleboard/beaglebone-black #3: Wrong connector number in SRM

smortex, 2017-12-06:
On page 85 of file `BBB_SRM.pdf`, Rev C.1, we can read:

* “# Both of these signals connect to pin 41 of P*11*.”
* “@ Both of these signals connect to pin 42 of P*11*.”

But there is no P*11* connector.  In fact, this is the section talking about P*9* so **P11** should be replaced by **P9**.

smortex, 2017-12-06:
Same paragraphs, “by removing R*221*.” and “by removing R*202*.”.

There are no R*202* / R*221*. As far as I understand, **R221** should be **R20**, and **R202** should be **R51**.  But one may prefer to remove R20 and R50 depending on the application… maybe it makes sense to tell the user about the 3 possibilities (use one pin as input, the other one as output ; remove one of the resistors ; remove the other resistor).

jadonk, 2018-01-17:
@smortex thanks!

Note that the latest SRM is now at https://github.com/beagleboard/beaglebone-black/wiki/System-Reference-Manual. I'm still working on the process of producing a suitable new .PDF file to replace BBB_SRM.pdf.

jadonk, 2018-01-18:
Updated on wiki: https://github.com/beagleboard/beaglebone-black/wiki/System-Reference-Manual/_compare/363d5c354752fb1a1734f4a465a1d885d3a89e16...0828a11554c69e32f2dd45afb77fd1157a17b44d

jadonk, 2018-01-18:
Note I've duplicated and expanded upon some of this table data at https://github.com/beagleboard/capes

jadonk, 2018-01-18:
OK, I think I've fixed it up enough.

closed 2018-01-18 by jadonk