beagleboard/beaglebone-black #6: P10 (SD Card Slot) Obsolete

abferm, 2018-11-05:
ALPS has discontinued SCHA5B0200 without an alternative or replacement. If there is a suitable replacement from a different manufacturer the BOM should be updated.

`P10	Connector SD/MMC	ALPS	SCHA5B0200`

> Dear Alex Ferm,
> 
>  
> 
> Thank you for your interest in Alps Electric.
> 
>  
> 
> With regard to your inquiry, we apologize but Alps discontinued all card connectors with no replacements or alternatives.
> 
>  
> 
>  
> 
> Best regards,
> 
>  
> 
> Lourdes Olanio
> 
> Sales Representative
> 
> Alps Electric (North America), Inc.

jadonk, 2020-01-10:
No suitable replacment has been found. They continue to supply this component. We did a design update with a new card cage, but it didn't pass FCC/CE and we currently have no plans to revisit. We continue to be able to design in high volume (~400k/yr).

jadonk, 2021-02-12:
Alternate proposed SD card cage options with associated layout update are:
* Molex 5027740891
* XUNPU TF-114

jadonk, 2021-08-30:
Resolved in rev_c3

closed 2021-08-30 by jadonk