beagleboard/beaglebone-black #12: Connecting the BeagleBone Black with mikroE Click Boards/mikroBus Cape help wanted

ckoehler0, 2021-06-18:
Hello,
I'm currently trying to connect my BeagleBone Black Rev C with the mikroBus Cape (PID: MIKROE-1857). 
Furthermore, I want to get access to a MRAM click module (https://www.mikroe.com/mram-click) by connecting it with the mikroBus Cape. My BBB is running the latest Debian image (AM3358 Debian 10.3 2020-04-06 4GB SD IoT) and with uname_r=4.14.108-ti-r141.

Firstly, I tried to work with U-Boot overlays:

1.
I followed the instructions on https://elinux.org/Tigal:BeagleBone_mikrobuscape but used the updated url "https://github.com/mikroe/BBB-mikroBUS-Cape". But, after setting the "uboot_overlay_addr0=/lib/firmware/BB-MIKROBUS-01-00A1.dtbo" in the uEnv.txt the board would not boot.

2.
I followed the instructions on "https://github.com/beagleboard/bb.org-overlays/" and I also upgraded my bootloader (https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays). After setting all four BB-MIKROBUS-CAPE-X.dtbo files in "uboot_overlay_addrY=..." the board booted. (It also boots when only one is enabled)

Next I wanted to get access to the click module on the mikroBus Cape. 

A.
I downloaded the mikroSDK 2.0 (https://www.mikroe.com/mikrosdk) on the BBB, but I couldn't find nor create the correct board.h file to get it running. After stumbling upon "NectoStudio" I could only find support for generic ARM boards which are from the STM32 family, which the BBB Rev C is not.  (Getting access to the mikroSDK 2.0 would allow the usage of scripts via https://libstock.mikroe.com/ which would open up many other possibilities)

As I couldn't get the mikroSDK 2.0 working, I tried to get direct access via SPI to the click module. For that I put the click module on the mikroBus Cape slot 3 (CS - Pin 17, D1/MISO - Pin 21, D0/MOSI - Pin 18 and CLK - Pin 22).
	
B. 
After some research I found the iobb library and followed its instructions 
(https://www.element14.com/community/community/designcenter/single-board-computers/next-genbeaglebone/blog/2019/08/15/beaglebone-black-bbb-io-gpio-spi-and-i2c-library-for-c-2019-edition).
After installation, I tried to run the example SPI code without any changes (so not using option 2.) and it runs error-free. But when I enable the BB-MIKROBUS-CAPE-3.dtbo I always get "ERROR: open() for /sys/devices/platform/ocp/ocp:P9_17_pinmux/state failed, No such file or directory"  So, after some investigation I realised that BB-MIKROBUS-CAPE-3.dtbo sets "P9_X_pinmux { status = "disabled"; };" for X = {18,19,21,22} and doesn't enable spi in its pinctrl-section, which could be the cause for the problem.

So I tried two things:
- I just used BB-MIKROBUS-CAPE-2.dtbo which doesn't set "P9_X_pinmux { status = "disabled"; };" for any X = {18,19,21,22} and the example code ran error-free.
- I changed BB-MIKROBUS-CAPE-3.dtbo so that spi gets enabled in the pinctrl-section and the example code ran error-free.
	
To my understanding both of this ways should enable SPI, but when I try to read previously written data from the click module, I can't receive any data at all or select a cell to write to.

So, is there a way to get any of my attempts working so that I can get access to the click module via SPI? Or have I overlooked an other solution?

Btw:
- I tried all solutions on the board itself and on a microSD card which I set up using https://beagleboard.org/getting-started
- I also tried it with and without setting "disable_uboot_overlay_video=1" in the uEnv.txt

jadonk, 2021-10-18:
Please ask for support on forum.beagleboard.org. If resolved, please close this issue and point to the discussion. If unable to resolve, please contact me via my calendar at https://beagleboard.org/about/jkridner.

ckoehler0, 2021-12-14:
The issues were due to an internal error of the beagle board.

closed 2021-12-14 by ckoehler0