beagleboard/beaglebone-black #16: TPS65217 not available anymore

mickae1, 2022-06-27:
Hi,

The TPS65217 is not available anymore, how can we replace it ?

cschweers, 2022-06-27:
According to TI the TPS65217 ist still [active](https://www.ti.com/product/TPS65217).

mickae1, 2022-06-27:
yes, but nowhere to be found !

cschweers, 2022-06-27:
> yes, but nowhere to be found !

... my distributor show me that 4.274 parts are available.

mvduin, 2022-06-27:
The entire semiconductor industry has availability problems, that's not something to be reported as a bug on a design.

You can use either the TPS65217C or TPS65217D, anything else would require a substantial redesign of the schematic and PCB and modifications to u-boot and the device tree.

mickae1, 2022-06-27:
I agree it's not a bug, but you can't built anymore beaglebone black if you don't have the component ...

Maybe it's time to create a new design ?