beagleboard/beaglebone-black #17: BeagleBone Black industrial can't login as debian

XuWei-BH, 2022-11-04:
The industrial version of BeagleBone Black can't be logged in as the debian or root account. The password can't be typed in.

XuWei-BH, 2022-11-04:
The image of "AM3358 Debian 10.3 2020-04-06 4GB SD IoT" and "AM3358 Debian 10.0 2019-07-07 4GB SD IoT " has this issue, it means the BeagleBone Black industrial can't been reboot while the "Debian 9.5 2018-10-07 4GB SD LXQT" can be logged in.

mvduin, 2022-11-04:
This is not an appropriate place for software support questions, this repository is for the hardware itself, issues should only be created here if they've been identified as hardware bugs.

However it sounds to me like you're not realising that when logging in in text mode linux doesn't show anything when you're typing in the password (which is how it traditionally has always worked). You can just type the password ("temppwd") when linux asks for it and press enter, it will work.

If you have any further trouble I recommend asking on the forum, https://forum.beagleboard.org/