beagleboard/beaglebone-black #19: Inconsistency between figure 39 and text in SRM + comments on the text relating to boot

TooManyTakenUsernames, 2024-01-11:
In Figure 39 of the user manual, first in the boot sequence when the boot switch is pressed, is "SPI0". 
The text states: "On boot, the processor will look for the SPIO0 port first". So the label between these is not consistent. 

Comment on the text:
Section 5.3.5 discusses 4 different boot modes, eMMC Boot, SD Boot, Serial Boot and USB boot. It could be helpful (especially for beginners) to relate back to these boot modes in section 6.8 (possibly in relation to figure 39). The text about the serial boot especially feels unconnected to section 6.8: "Serial Boot…This mode will use the serial port to allow downloading of the software direct. A separate USB to serial cable is required to use this port." 

I'm assuming that booting from SPI0 or UART is booting in the serial mode, but I cannot confirm this anywhere just by reading the text.

mvduin, 2024-01-11:
Issues for the [current SRM](https://docs.beagleboard.org/latest/boards/beaglebone/black/) should probably be reported [here](https://openbeagle.org/docs/docs.beagleboard.io/-/issues) instead.

It looks like the [current version](https://docs.beagleboard.org/latest/boards/beaglebone/black/ch05.html#boot-modes) doesn't even mention SPI boot, which I personally don't think is an improvement since even if you never use it, its presence in the boot sequence (when the boot switch was down at power-on) means P9.17, P9.18, and P9.22 will be configured as outputs and will toggle as a result of bootrom's attempt to boot from an SPI flash on SPI0 CS0.

Serial boot is referring to UART0 (XMODEM) boot only.

TooManyTakenUsernames, 2024-01-12:
Thank you! And many thanks for clarifying that the serial boot refers to UART only. 
I was reading the [PDF version](https://github.com/beagleboard/beaglebone-black/blob/master/BBB_SRM.pdf) and the link for supplying comments and errors in there leads to here:

"Supply comments and errors via https://github.com/beagleboard/beaglebone-black/issues" 

I will close this issue and open an issue at the link you supplied instead.