beagleboard/beaglebone-black #20: Conversion of (0-5V) Analog Signal to (0-1.8V) without Loss of Information

NabilaTasnim, 2024-04-14:
I have an analog signal of (0-5V). I want to use the ADC of BeagleBone Black to process it, but the ADC inputs can take a maximum of 1.8V. If I down-scale the signal to (0-1.8V), then -

How should I do this ensuring no loss/minimum loss of information or resolution? If there is some loss, then is there any way to retrieve the original signal after processing it with the ADC of BeagleBone Black?

cschweers, 2024-04-14:
This is precisely the purpose of the so-called [voltage divider](https://en.wikipedia.org/wiki/Voltage_divider) in [electrical engineering](https://electronics.stackexchange.com/)

NabilaTasnim, 2024-04-14:
If I use a voltage divider to scale down the voltage and then use ADC for digital conversion, won't I lose some accuracy and information? @cschweers @jadonk @RobertCNelson @pdp7

jadonk, 2024-04-16:
yes, but not much depending on the accuracy of the resistors and amount of capacitance on the line.

mvduin, 2024-04-16:
No, scaling the full range of the signal you want to measure to match the full range of the ADC you're using to measure it is exactly how you achieve the best accuracy possible with that particular ADC.

Also, this sort of question is better asked on the forum at https://forum.beagleboard.org/ instead of using a github issue.