beagleboard/beaglebone-black #21: BeagleBone Cape Board Design without EEPROM

NabilaTasnim, 2024-05-07:
I am confused about one thing. If I design a BeagleBone Cape Board without EEPROM and use it with BeagleBone Black, then will that Cape Board work just like a PCB board? @RobertCNelson @jadonk @pdp7 @MarkAYoder

RobertCNelson, 2024-05-07:
If you don't install an eeprom, it just won't auto-load... Nothing stopping from writing directions on using an overlay for your cape.

NabilaTasnim, 2024-05-07:
@RobertCNelson I was thinking if I connect the cape board over the BeagleBone Black and control the GPIO pins of the BeagleBone Black, will I be able to use the cape board? I meant, can I then use the cape board just like the breadboard or protoboard?

RobertCNelson, 2024-05-07:
eeprom is only used for 'auto' detection of what board is plugged into the bbb.. as long as you know what board is plugged in your can load the appropriate overlay

NabilaTasnim, 2024-05-07:
But I want the cape to serve the purpose of the PCB board. It does not need to be intelligent. The cape’s header pins will be like jumper wires, so if I enable one GPIO pin of BeagleBone Black, the corresponding header pin of the connected cape will also be enabled like the jumper wire. So, is it possible to use the header pins of the cape like the jumper wires? @RobertCNelson

RobertCNelson, 2024-05-07:
The only intelligent feature is the erprom. If you don't have eeprom, you can write your own overlay to configure the pin in any mode that is supported.

adhishhh, 2024-06-18:
Yes, your Cape Board without EEPROM will work like a PCB with BeagleBone Black. You'll just need to create a Device Tree Overlay (DTO) for manual configuration instead of automatic detection via EEPROM. This lets you control the Cape's functionalities through BeagleBone Black's software, similar to how you'd use jumper wires on a breadboard.